wled/WLED #872: WLED color changing not possible

WS2812BNodeMCU, 2020-04-25:
Hello and sorry in advance for my bad English. 
I installed the latest WLED binary on my NodeMcu but when I pick a color on the color wheel, the LEDs change their colors to my picked but they instantly switch back to the previous color.
Sometimes the picked color then come up after like 10 seconds but most of the time the color does not change at all. 

I already tested a lot. Switched time server, tried an older binary, only use one led, disable wifi sleep etc. 
I use a NodeMcu and an external power supply. The problem also exists when I use USB power.

What else I can check? Thank you.

raoulteeuwen, 2020-04-25:
Strange. Do you now the exact brand/name/type of the controller?

1. You already tried a complete NodeMCU wipe ("esptool.py erase_flash") & new flash? 
2. Do you happen to be able to try with a new controller (some 8266 device)?

WS2812BNodeMCU, 2020-04-25:
> Strange. Do you now the exact brand/name/type of the controller?
> 
> 1. You already tried a complete NodeMCU wipe ("esptool.py erase_flash") & new flash?
> 2. Do you happen to be able to try with a new controller (some 8266 device)?

I will try both later. Thank you very much.

Aircoookie, 2020-04-25:
In addition to that, could you try disabling the WLED notifier sync in settings? There is a (low) possibility that there are some packets on your network that wled mistakes as sync packets and "syncs" to an off state...

closed 2020-08-30 by stale[bot]