wled/WLED #873: Color wheel changes to black after config

EvertJob, 2020-04-25:
Hi,

Just installed WLED on a NodeMCU and everything’s works fine as far i can see right now (LED’s should come in today) but after visiting the config page the color wheel is black.

When I re-open a private window and visit the WLED page the color’s are back and the same issue is happening after visiting the config page.

Tried a second NodeMCU: same issue. As far i don’t open the config page the colors are there.

Any one a suggestion to fix it?

AlroyLED, 2020-04-25:
I have same issue on multiple devices.

raoulteeuwen, 2020-04-25:
@EvertJob what exact bin (or build from master or something else) did you use?

EvertJob, 2020-04-25:
Server message: WLED 0.9.1 (build 2002222) OK

I’ve ordered 6 SK6812 strips from 60 leds each for under the stairs. The idea is to use it with WLED in HomeKit but the Domoticz route it not working and also the Homebridge route is not working. Anyone a suggestion for that? The other option is to use Tasmota

raoulteeuwen, 2020-04-25:
So you downloaded a BIN from https://github.com/Aircoookie/WLED/releases ? "WLED_0.9.1_ESP8266.bin"?

Does that 'darkening of the color wheel' happen both when in AP-mode as well as when connected to your WLAN? Just to see if we can rule out connection issues...

srg74, 2020-04-25:
Since 0.90b color wheel become black when you return from settings. This issue have been reported before. Happening in Safari on OS X and iOS devices and WLED iOS app.

Aircoookie, 2020-04-25:
Interesting. Supposedly this was fixed in v4 of the iro.js colorpicker library, but perhaps not in some edge cases like loading the page from cache... Will need to check it out, maybe switching to iro.vs v5 could help.

EvertJob, 2020-04-26:
> Interesting. Supposedly this was fixed in v4 of the iro.js colorpicker library, but perhaps not in some edge cases like loading the page from cache... Will need to check it out, maybe switching to iro.vs v5 could help.

> Interesting. Supposedly this was fixed in v4 of the iro.js colorpicker library, but perhaps not in some edge cases like loading the page from cache... Will need to check it out, maybe switching to iro.vs v5 could help.

Strange because this is only happening when visiting the config section.

kristof-dragon, 2020-10-09:
I have the same issue with the app on iOS 14.2 using ESP8266s with the v0.10.2 .bin upoloaded to them.

Backing out from the Setting makes the colourwheel go black. (although you can still use it to select the actual colour)

Rstarting the App or backing out all the way to the main screen (where you can select the configured WLED modules) and opening up the same device resolves the issue...

danuals, 2020-10-31:
iOS 14.2 app and web page the same. 
[image]

fgonza2, 2020-11-20:
same issue here

Aircoookie, 2020-11-20:
This is now fixed in the latest master source code and will make its way into release 0.11.0 on Sunday :)

closed 2022-10-21 by Aircoookie