wled/WLED #874: 2 different stripes different colors, can this be adjusted

StefanRu1, 2020-04-25:
Hi,

today i got a new 5 MT WS2812B.
As i extended one of my stripes with the new one i recognized that the new band is more yellow as the old one. Or the old one is more red.
I recognize the difference most when i use a orange color.
With the other colors (Blue, Green) i do not recognize a big difference.

Now i installed the new one in the kitchen and the old one in the room next to the kitchen.
But the not matching colors bothers me.
Is there a way to adjust this somehow?

Thanks and best regards,
Stefan

Aircoookie, 2020-04-25:
Hi,

firstly make sure both strips receive enough voltage - make sure the power supply is adequate and there are enough power injection points.
Other than that, it often happens that strips from different sellers or charges have different colors - these are unfortunately not very uniform in that regard. I might add a color correction option at some point :)

StefanRu1, 2020-04-25:
Hi Christian,

yes booth stripes have the same power source and enough voltage.
I also soldered a piece new and a piece old stripe together. 
The colors are at least different in the yellow to red tones.

In my home automation, which does a yellow to red slide depending on the time of the evening, i could correct it by removing 20 from the green value for the new stripe.

But this only helps when the light is driven from my home automation. 
If i control it with the WLED UI i have the color difference.

It would be great if there would be a color correction option in WLED.
If i can help somehow, e.g. with testing, i would be happy.

Thanks and best regards,
Stefan

pille, 2020-09-01:
please reopen to track this color correction feature request.

StefanRu1, 2020-09-01:
Ok, will reopen it.
Thanks alot for the reminder and the new release "Fumikiri" it works great.

StefanRu1, 2020-09-01:
Oh i am sorry i can not reopen, because i did not close it.. The bot did it..

closed 2022-02-21 by blazoncek