wled/WLED #875: Shows unavailable when connected to home assistan

comangus13, 2020-04-27:
Hi, thanks for all your efforts with WLED. I have updated to version 0.9.1 and works well by itself (WLED APP or Web Browser), however, every time I connect the integration on my home assistant (currently version 0.108.9) start disconnecting and connecting all the time, showing it as "unavailable". I have tried on all the 0.108 versions

You can see from the timeline (see attachement) that fluctuates all the time. Also, I get this error from the WLED app (see attachement)

Once I disconnect the integration from HA, it gets back to normal without issues

Any advice?

Let me know if you need further information

Cheers,
Gus

[image]
[image]

austwhite, 2020-04-27:
This is a Home Assistant integration issue rather than WLED.
I think you may need to log a call against the plugin for Home Assistant as this is written by a different person.

@Aircoookie if you have any ideas, let me know as I also have similar issues.

austwhite, 2020-04-27:
@comangus13 
Just one other thing.Did you try manually configuring it as an MQTT light?
I did that once and it worked well with no disconnects.
I generally only use my WLEDs for occasion lights so they don't stay up all the time so I haven't got the MQTT YAML to hand right now

raoulteeuwen, 2020-04-27:
Also see https://github.com/Aircoookie/WLED/issues/843 . Maybe close this ticket if you think it is a duplicate.

comangus13, 2020-04-28:
@austwhite Thanks for the info, I just added it muanually with MQTT and works well. 
 I will create a ticket with HA to see if I can use their WLED integration

@raoulteeuwen, sure man np

closed 2020-09-03 by stale[bot]