wled/WLED #879: Issuing commands too quickly causes boot loop

L337Lobster, 2020-04-29:
Version: 0.9.1
Board: [NodeMCU (i believe 12-E. Amica brand)](https://smile.amazon.com/gp/product/B07B63ZYM1/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)
Lights: [12v SK2812](https://www.aliexpress.com/item/33018762802.html?spm=a2g0s.9042311.0.0.66a14c4d9GjpEz)
Power Supply: [12v 30A](https://smile.amazon.com/gp/product/B06XJVYDDW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
DC to DC power supply module: [Adjustable](https://smile.amazon.com/gp/product/B078Q1624B/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
)
Issue: I've been have multiple issues, some have been fixed through trial and error. But they all seem to relate to connection. I have a similar issue to #813 occasinally. But the main issue I've noticed is it will go into a boot loop if I accidentally issue commands too quickly. I've had them up and running over night before and then I'll wake up and have the #813 issue where I can see it connected to my router but the UI won't load. Thinking of how my MQTT is setup that could simply be caused by too many commands too quickly.

Long story short I have a resistor between 3v3 and D4 (GPIO2) and a shared ground between the lights and the unit. This initially solved my boot loop issue as it seemed to be caused by pulling GPIO2 low, but since then it still get's thrown into a boot loop that I can't get it out of without a full reflash of the firmware. A reflash doens't always seem to fix it however. I'm not sure what else I can provide information wise. This last time it got into a boot loop it connected when plugged into my computer then I put it back into it's slot and it wouldn't connect even when I disconnected my lights from GPIO2. I'm very new to this and learning as I go so please bear with me if I'm overlooking something simple or simply not providing enough information.

huggy-d1, 2020-04-29:
There are people out there trying to solve this same (or _very_ similar) problem with wifi dropping.  Someone posted a code change that worked for them.  Something about a well placed wifi.disconnect call magically fixed their issues.

L337Lobster, 2020-04-30:
I believe it's related to MQTT. My timer was sending an on command once every minute and once I disabled that and reflashed it the issue appears to have gone away for now. Will update if that changes. I reenabled MQTT but only for Openhab control, not on a timer outside of the built in one. Hopefully that fixes it.

triggerx, 2020-05-06:
> I believe it's related to MQTT. My timer was sending an on command once every minute and once I disabled that and reflashed it the issue appears to have gone away for now. Will update if that [...]

Try disabling Alexa in Sync Interfaces and see if that helps.

closed 2020-09-10 by stale[bot]