wled/WLED #882: JSON often not applying command

chiefymuc, 2020-04-30:
WLED 0.9.1 on a WEMOS D1

I'm controlling WLED through the JSON HTTP API and I am seeing an inconsistent behaviour when sending commands. 

I always get the answer {"success": true} to my POST requests. However, often (I would say 80% of the time) the command does not get applied.

Interesting is that if I wait a bit, maybe 20-60s, the next command has a higher chance of success, although not always.

How is this possible when I get the "success" response?

It is the exact same command that I am sending to make sure there are no wrong commands...

Aircoookie, 2020-05-04:
Hi!
What is the exact command you are sending? It maybe that no Segments are selected, so nothing appears to happen even though the command is received correctly. You can also add `"v":true` to your command, it will then respond with the state object and you can see if the value changed :)

chiefymuc, 2020-05-05:
I investigated some more and it seemed like my script sent the wrong value. My bad. Closed.

closed 2020-05-05 by chiefymuc