wled/WLED #883: 0.9.1 sending SYNC when disabled

nicr4wks, 2020-05-01:
Hello,

I've just flashed a new NodeMCU with 0.9.1 binary.

I have all send sync options disabled:
https://imgur.com/qFK9O8n

Randomly this WLED instance will start controlling all of my other installations, if I turn the lights ON, all installations turn on, they all change color, they all change brightness etc.

I visit the SYNC SETTINGS page and all tick boxes are showing disabled, I click save without changing any settings and it stops sending commands to my other installations. I've had this new installation running for roughly 24hrs and this has happened 3 times.

Thanks.

Aircoookie, 2020-05-02:
Hi!
It seems like sometimes the setting is only applied after rebooting. Also please make sure that the Sync button isn't lit in the web UI. If nothing else helps, you can change the UDP port and reboot.
Sorry for the issue and thank you for using WLED :)

Morphy99, 2020-05-02:
I've had this issue with sync switching on after a reboot. It only happens with one of my ESP's, it's a NodeMCU. It's also one I've had the longest so would have been on v0.84 originally. All my other Wemos D1 Mini's don't do this but they would've started on a later version. Not sure if any of that is relevant or if it's just a setting I'm missing?

nicr4wks, 2020-05-03:
I's been a few days now and the issue hasn't reappeared, hopefully just a small hiccup.

closed 2020-09-08 by stale[bot]