wled/WLED #884: How to switch between usermods

francescofact, 2020-05-01:
Hi, I would like to jump into wled and start developing some games or other things for who has a wled as 16x16 matrix driver.
I don’t get some things and if this project is what I’m looking for.
- can I add more than one usermodule? 
- how can I switch to the module?
For example If I have three modules, like FlappyBird, Clock, Weather, how can I tell the wled to switch between them?
Thank you

mike2nl, 2020-05-02:
@francescofact 
if you wish to use a different usermod ()FlappyBird, clock, ... you have to rename the usermod file to the one in the WLED folder, then compile it and download it. That;s how it works now.

i was thinking about an interface which can call differentr mods from the web ui but that is very complex. Had made some very small tests but it is far, far away from working. 

At the end @Aircoookie is the project owner and he has the last words and he knows the code at best. At the moment there are no discussions about that heavy development.

huggy-d1, 2020-05-08:
The simple solution would be to have your (currently single) usermod accept button input.  Each button press cycles through whatever modules are enabled in the usermod.  Of course, at least one of those options should yield control to the user interface, and that would be the module in your usermod that simply returns, doing nothing other than look for another button press.

mike2nl, 2020-05-09:
@huggy-d1

closed 2020-09-13 by stale[bot]