wled/WLED #886: mqtt with alexa command

sjude68, 2020-05-02:
@Aircoookie How do I configure sending MQTT msg when Alexa voice command is given turn ON or OFF the Led stip.  I also added 4-relays with MQTT.  I can use Alexa commands but I need to send the MQTT msg so it can be monitored.

Help me modify the code.

Aircoookie, 2020-05-05:
Did you add your relay topics to the `publishMqtt()` function? if yes, it should be enough to set `doPublishMqtt = true;` in the Alexa relay callback function. If you made your own mqtt sender, you will have to add a bool to activate it when the relay was changed via Alexa.

sjude68, 2020-05-06:
Thanks @Aircoookie. I added mqtt->publish lines after receiving the topic and then `doPublishMqtt = true`. I get the Mqtt msgs after Alexa voice commands. But not when the main wled light strip is ON/Off. Only the bri is updated from `0`.

Now that you have released the new version, I tried moving the usermod lines to the new version but couldn't succeed. Also I had to use pin numbers instead of D5, D6, etc.

sjude68, 2020-05-26:
I implemented the modified code in New version. It works for 4ch relays with mqtt publish.

sjude68, 2020-07-06:
In version 0.10.0, I am not able to use publish statement in usermod.ino 

`mqtt->publish("home/hall/temperature", 0, true, temperatureTemp);`

It causes the Nodemcu to reset at that statement. I am reading temperatue from a DHT sensor and publishing to the local MQTT server which is already setup in WLED configuration.

When I used `if (WLED_CONNECTED)` and used the publish within the if loop there is no reset but no publishing to mqtt server.

Aircoookie, 2020-07-10:
Hi,
interesting that that doesn't work, as soon as the MQTT server is connected, the publish function should work.
You can try if it works using `if (mqtt != nullptr && mqtt->connected())`, which will specifically check for MQTT and not only general WiFi connection.

sjude68, 2020-07-10:
Hi @Aircoookie ,

It does not work with the line suggested.  I have attached the usermod.cpp file for your reference.
[usermod.zip](https://github.com/Aircoookie/WLED/files/4905073/usermod.zip)

sjude68, 2020-07-10:
sorry I had to wait for 30 minutes to get the publish because I used an interval of 30 minutes.  when I reduced for 1 minute I got the publish on the 2nd loop.

Aircoookie, 2020-07-11:
Ah, good that that was the issue!
You can remove the `time_now` in line 17 as you only need it as a local variable.
Another small tip if you want a larger interval, but for it to update soon after boot is to set initialize previous_time as `unsigned long previous_time = `(UINT32_MAX - inter) +60000;` which will publish once a minute after boot and then with the interval specified :)

sjude68, 2020-07-11:
Thanks @Aircoookie

closed 2020-11-16 by stale[bot]