wled/WLED #890: WLED UDP

DennisH1978, 2020-05-03:
Hallo, ich schreibe euch ausnahmsweise mal auf Deutsch, weil ich gar nicht genau weiß, wie ich es sonst übersetzen sollte. Ich habe derzeit 9 NodeMCU mit WLED drauf, super Sache, bin begeistert. Da ich aber die 7 NodeMCU im Wohnzimmer unabhängig von denen draußen (2 Stück) steuern möchte, habe ich mich entschlossen, den 7 Stück im Wohnzimmer den UDP Port 1 zu geben und den beiden draußen den UDP Port 2 (alles Port 1884, da auf 1883 schon sehr viel drauf ist). So dachte ich, erreiche ich 2 Gruppen, dich ich entweder über die Android App oder MQTT im IoBroker getrennt voneinander steuern kann. Funktioniert tatsächlich auch. Nur habe ich gerade gelesen, dass das wohl gar nicht für meine Zwecke gedacht ist?! Dazu ein weiteres Problem: Die 7 NodeMCU aus dem Wohnzimmer greifen direkt auf meinen Router in der Nähe des Wohnzimnmers zu, die anderen beiden auf den Repeater, der im Mesh Modus läuft. Sitze ich im Wohnzimmer und habe die App offen, zeigt er mir häufig an, dass die beiden außen offline sind. Stehe ich draußen, ist es häufig umgekehrt. Hier werden die 7 im Wohnzimmer als offline angezeigt. Im Grunde kann es ja nicht an WLAN liegen, da bei Mesh ja alles dieselben Zugangsdaten hat. Vielleicht ist es aber auch ein einfacher Fehler, an dem ich jetzt vorbei sehe. Was meint ihr? Und wie bekomme ich das am besten mit den Gruppen hin?

Aircoookie, 2020-05-05:
Hi :) Super dass dir WLED so gut gefällt!
Verschiedene UDP Ports sind schon ein guter Weg, verschiedene Kontrollgruppen zu erzeugen. Ich würde aber Portnummern über 20000 empfehlen, alles darunter ist oft für andere Protokolle reserviert. Wegen des Repeater-Mesh-Setups, dafür ist es nicht maßgeblich, ob die WLAN's die gleichen Zugangsdaten haben, sondern ob sie im gleichen Subnet sind. Einige Repeater haben eine NAT, da können die Geräte am Repeater die im anderen Netz nicht erreichen und umgekehrt. Kann man meistens in der Konfiguration einstellen :)

DennisH1978, 2020-05-06:
Hallo Aircookie,

vielen Dank für deine ausführliche Antwort.
Meine FritzBox nutzt nur das eine Netz. Subnetze habe ich nicht, zumindest
nicht eingerichtet. Ich werde dann Portnummern über 20000 nutzen und
schauen ob es besser wird.
Merci :-)

closed 2020-09-11 by stale[bot]