wled/WLED #891: Segments are lost after each reboot or power cycle

ikrowni, 2020-05-04:
Everytime i turn off the power supplt or reboot the esp8266 my segments are deleted and i have to set them back up. Is there anyway to fix this?

tube0013, 2020-05-04:
Save them to preset 16. That is the only way to save segments currently.

ikrowni, 2020-05-04:
trying to save to slot 16 it says "saving config to slot 16" but then when i uncheck saving mode and click on slot 16 it says "Slot 16 is empty. Use saving mode to save the current config to it" I've tried over and over again. tried rebooting the sp over and over. just doesnt seem to want to save it for some reason. This problem started upon upgrading tot eh latest version 0.10.0

Legsmaniac, 2020-05-04:
@ikrowni Thank you for asking this question and @tube0013 thank you for that answer, I didn't know preset 16 did that.  I had set up my segments using the userMod file from source and recompiled.  It worked but having just tried saving to preset 16, it was a cleaner and easier (and faster) job so I've deleted my userMod script and now using preset 16.

@ikrowni - I'm using version 10 and can confirm it works well.  Set up segments, modes and colours etc. as required, click on Favorites, check Saving Mode and click 16.  Then uncheck saving mode.
Now click the cog to take you to settings. (Config)  Click LED Preferences, scroll down to Apply preset at boot.  Type 16 in the box then click Save button top or bottom of page.
Turning on your device now sets Preset 16 every time.

@Aircoookie Hey, I finally figured Git and made it on here!  Hi!  (Waves)  :-)

Aircoookie, 2020-05-04:
Hi! :)
I just confirmed this! Darn I cannot release a version without discovering a bug immediately after. The preset 16 indicator always indicate as no preset being saved, even though it is actually saved and restored correctly. This is just a visual bug, albeit a very misleading one. Rest assured that preset 16 is working normally :) Will be fixed in source later today!

closed 2020-05-04 by ikrowni

mike2nl, 2020-05-09:
> Hi! :)
> I just confirmed this! Darn I cannot release a version without discovering a bug immediately after. 

@Aircoookie 
That is 100% normal, believe me. I have my experinece with that. You know from where.

austwhite, 2020-05-09:
@Aircoookie definitely normal to find bugs after release.  Microsoft are king at that. Take a look at how many updates Home Assistant gets too.  I think the last one get 9 patches

ikrowni, 2020-05-09:
@Aircoookie are there plans to allow segments to be saved to more slots in the future?

Aircoookie, 2020-05-09:
@ikrowni Currently working on a system that stores the presets in the built-in filesystem. No promises yet, but we should be able to support hundreds of multi-segment presets :D

ikrowni, 2020-05-09:
Beautiful!

commit 329b2ba2 (2020-05-08): Fixed Preset Slot 16 always indicating as empty (fixes #891)