wled/WLED #892: OTA Updating to 0.10 fails during upload

fribse, 2020-05-04:
Hi Aircookie

I can't wait to get the 0.10 update on the lights, especially the detail with the multi-candle was great news for my small projects (#541) :-)
I also saw that you included the info page for future updates, which is highly appreciated here (#750)!
I have a mix of D1 mini's and ESP32's in D1 mini format mounted on diguno boards around the house.
So far, no matter if I try to flash from the laptop on wireless, or from the computer on cable, I fail to update the esp's. That is both esp8266 and esp32's. They are all running 0.91 at the moment.
I've used your hint with /json/info to make sure if it's a 8266 or 32, so that is not the issue here.
I've tried both from Firefox and Chrome (on win10).
Chrome gives me:
```
This page isn’t working
10.11.13.115 didn’t send any data.
ERR_EMPTY_RESPONSE
```
Firefox gives me:

The connection was reset

I'm not sure what is going on behind the scenes.
When they fail, and I press F5, half the times I see 'update completed successfully wait for reboot' the other half 'update failed'. No matter what it shows 0.91 after the process.

I've found a procedure that works, but it's quite cumbersome.

1. Note down as much as I can from the config (loosing the presets)
2. Reset the wled to factory defaults.
3. Connect to WLED-AP
4. Update wled while in defaults
5. Set the right SSID (seperate VLAN)
6. Set all the config up again

Aircoookie, 2020-05-04:
Hi, I'm sorry that you ran into issues while updating... Did you check whether the WLED-AP is active when you could no longer connect to the IP? Do you know the exact binaries you used for 0.10.0 and 0.9.1 respectively? If I understood you correctly, your boards went back online after the failed update, just still running version 0.9.1... It might also be that the upload is interrupted due to poor reception - you could try uploading closer to the router (or check if changing the VLAN of your uploading device works, albeit i doubt it)

fribse, 2020-05-04:
The AP wasn't active.
The versions was the regular for pin16 (diguno).
Yes, they went back online, but gave difference messages on success or failure.
I doubt it's a reception problem, because it worked fine after the factory reset.
I also tried moving them to another VLAN (the main one), but no change.

fribse, 2020-05-04:
I also tried 'updating' from one 0.10 to 0.10, but it fails. I guess I could just swap it to a esp32, but it seems overkill for 59 leds on this one :-)

bartekd123, 2020-05-05:
MIne kept failing too. I put it close to the wifi access point and still had the issue. Always 500 error. Tried different browsers, and a different computer. 

I ended up flashing with a USB cable, and re-doing my settings.

jezmck, 2020-05-05:
I too had a failure when doing OTA to 0.10, not sure what information would be helpful, lmk if I can help.

bphick, 2020-05-28:
Hi.  I tried to do an update from WLED 0.8.6 to WLED 0.10.0 via OTA on 2 of my ESP32 NodeMCUs.  From the phone, pressing the update at times would cause the lights to briefly flash white, but no update happened (Android 9.0 on a Samsung Galaxy S8 phone). After trying that many times, I gave up and moved to a PC to try.  The update SEEMED to go OK (Windows 10). All info was at the bottom left on screen showing progress.  Said complete, then went to reboot... wait for reboot.  Then IP not accessible.  Power "blue light" on ESP no longer lit.  First 2 LEDs on the strip went on, one white, second green.  I thought I did something wrong...  Tried it again on MCU #2. Same result.  Cannot communicate to the board.  Tried manual USB connection.  PC does recognize port as open, but nothing will communicate with either of the NodeMCUs anymore.  The power light just flashes at connection, and goes out.  I tried to upload a blink script from the Arduino IDE and still fails to communicate...  I'm kinda thinking that my boards are fried.

closed 2020-10-03 by stale[bot]