wled/WLED #893: One ESP8266 gives different errors 'TypeError: info is undefined'

fribse, 2020-05-04:
Hi Aircookie

So one of my wled's doesn't want to do anything, I can't OTA at all, not even a reset and restore.
So I tried flashing it via tazmotizer and that worked.
After it's up and connected I'ved seen different errors from it.
After flashing: 

> WLED error 2

When trying to turn on the light: 

> TypeError: info is undefined (doesn't turn on the light)

After modifying the config: 

> TypeError: json is null

Is it just a bad esp8266?

fribse, 2020-05-04:
Ok, I've unwraped an ESP8266 to replace the other one, that works, it reacts rather slowly on ON/OFF commands.
I still see 

> WLED error 2

 on the UI

Aircoookie, 2020-05-04:
Hi :)
WLED error 2 means that saving settings to flash memory has failed. This means that the bin file flash layout is different from the hardware layout. Have you maybe used `WLED_0.10.0_ESP8266_1M.bin` instead of  `WLED_0.10.0_ESP8266.bin` (the correct one for Wemos D1)?

fribse, 2020-05-04:
No I used the regular esp8266 bin, the d1 mini is from geekreit as far as I recall it should be fully compatible with wemos.

fribse, 2020-05-04:
Ok, I've used tazomitizer, but if I switch to esphome_flasher, and then flashes it, it doesn't show the WLED error 2?
But it's still extremely laggy, and unstable, it takes 1-3 s for it to turn on and off.
I see messages with 'lost connection with esp, if it happens again reboot' or something like that.
It could be some strange combination with the wifi (unifi), but it's odd that it only happens for the esp8266.
For now I've just swapped it to an esp32, not a big deal, just odd.

triggerx, 2020-05-06:
> Ok, I've used tazomitizer, but if I switch to esphome_flasher, and then flashes it, it doesn't show the WLED error 2?
> But it's still extremely laggy, and unstable, it takes 1-3 s for it to turn on [...]

Try disabling Alexa in the Sync Interfaces and see if that helps.

fribse, 2020-05-06:
Everything in Sync is turned off :-) I always do that to save CPU power.

huggy-d1, 2020-05-08:
I saw in one of the issues or discourse topics where someone was getting strange behavior because their HA software was frequently capturing status (or similar) information from WLED.  If your HA system is doing that, perhaps try disabling it, or extending the time between polls.

fribse, 2020-05-10:
Thanks @huggy-d1 I have integrated it into HA, with the normal HA integration, and I have no idea how to do that :-)
All the problems went away when I swapped it for an ESP32, so I do beleive it's an overload situation. It's rather silly to have an ESP32 handling 59 LED's, but for now at least, there is peace on the home front.

fribse, 2020-09-07:
The problem is still at 0.10.0, I haven't upgraded yet.

Aircoookie, 2020-09-07:
Alright, let me know once you have more info :)
It just came to me that this might also be a client side issue - what browser do you use? Then again, if it was a browser issue, the same error should occur with ESP32, so I suspect a memory problem or something along those lines.

fribse, 2020-09-08:
I've updated yesterday to 0.10.2, I'll try to force the error to see if it's solved.

fribse, 2020-09-13:
I haven't seen the error on 0.10.2 yet, I'll close this, and if I see it again, I'll reopen.

closed 2020-09-13 by fribse