wled/WLED #894: 0.10.0 Nodemcu Segments

abyssdj, 2020-05-04:
Afternoon. I'm finding that my segments (2x 60) are not saving. They work correctly when set, but when powered off or rebooted, the segments disapear and reset to 1x 120 string. I don't think I'm missing a setting anywhere and the issue has been present in other versions too.

abyssdj, 2020-05-04:
I managed to "fix" this by selecting both current cycle and also apply 16 on startup options in the led settings. Is this correct?

Peterabboud, 2020-05-04:
I am having the same issue, but i don't want to use 16 on startup as 16 is the only one where you can use multiple segments so i want to keep it for a nice multicolor preset, but want the light to always start on warm white. Any workaround for that ? Also i just finished building my light today and owe a lot of gratitude to you Aircookie, as you have saved me possibly months of trying to figure out how to code this :D
[image]

jes1417, 2020-05-04:
@Peterabboud what did you use to build the frame?

Peterabboud, 2020-05-04:
I ordered an aluminum angle (15.8mmx15.8mm/1.5m length) and two aluminum channels (15.8mmx15.8mmx15.8mm / 0.5m length). In the front of the frame i stuck 3x50cm LED diffusers (L shaped with a curbed diffuser). The channels are good so you can pass the cable under it. I used JB weld to glue the channels to the main angle, but i would highly recommend to go and get them welded, as the JB Weld is a little fragile unless you put copious amounts of it. The light is a great success and is amazing when put in a corner.

jes1417, 2020-05-04:
It looks amazing

Peterabboud, 2020-05-04:
@jes1417 Links to build it ? I do not really have one, I just saw the original one which was being sold for 180£ (appaling) : https://minimalamp.com/products/minimal-lamp-vibrancy ,  and then i brainstormed/searched for parts for a day or two and set it up.  The whole thing cost me less than half of that price and took me a 2-3 days of assembly. I might write something up, but really its very straightforward :

2 floor channels, glued to the  outside of the aluminum angle
3 LED diffusers (L shaped) glued to the inside angle
1.5m of 144/m ws2812b LEDs (I bought two 1m strips, cut the second one and soldered it)
1 10A power supply
1 Wemos D1 mini (which i glued to the power supply)

[image]

jes1417, 2020-05-04:
I meant links to the parts

Aircoookie, 2020-05-05:
@Peterabboud that looks awesome! If you want pure warm white startup, just save that in a different preset and set it as boot preset. On boot, only one segment is activated over the full lenght of the strip. And brace yourselves, WLED 0.11.0 will give ya 50+ preset slots that all support segments!

Aircoookie, 2020-05-05:
@jes1417 you can probably buy similar stuff in every hardware store :) or just search on ebay for it.

@abyssdj yep, the preset 16 "fix" is the only way to currently persist segments. Will be improved soon!

Peterabboud, 2020-05-05:
@Aircoookie That's amazing! cant wait for the release :D Thanks again for all your hard work, i'll make sure to contribute once I am able to.

closed 2020-09-09 by stale[bot]