wled/WLED #895: 0.10.0 Nodemcu preset cycle

abyssdj, 2020-05-04:
Afternoon once again. In addition to the other bug I filed, I am finding that the preset cycle is overriding the software power switch and nightlight timer. Both work, but output is switched back on when the preset changes every x seconds. Is this expected behaviour, or am I missing something?

abyssdj, 2020-05-04:
I managed to "fix" this by selecting both current cycle and also apply 16 on startup options in the led settings. Is this correct?

Aircoookie, 2020-05-09:
Hi! This behavior is intended, but I agree that it does not make a lot of practical sense, so I believe it would be a good change to have the power button take priority over the preset cycle :)

abyssdj, 2020-05-18:
> 
> 
> Hi! This behavior is intended, but I agree that it does not make a lot of practical sense, so I believe it would be a good change to have the power button take priority over the preset cycle :)

Either that, or have a setting so that both options are available. I can't think of a practical reason for the cycle to override the power button, though.

SteveEdson, 2020-06-08:
FWIW I'd definitely prefer for the power button to override the cycles by default. I turn the WLED off via Home Assistant, and WLED is turning them back on via the cycle.

(Probably a thing to raise in the HomeAssistant repo, but it'd be cool if there was a way to control the presets via HA too)

closed 2020-09-04 by Aircoookie