wled/WLED #896: relay help

dnaoci4414, 2020-05-05:
Hello, I am writing to you from Mexico. Excuse me if I have any error in writing. I love wled. I made a chandelier for my room and agree 110v bulb lihgt,  I control wled it with Alexa, I manage 110v lights with a relay, my question is how can I connect it with the same card nodemcu v3 and be able to tell alexa turn off the light

[image]

raoulteeuwen, 2020-05-05:
Hi. Did you find this info regarding using relays: https://github.com/Aircoookie/WLED/search?q=alexa&type=Wikis ?

And since i do not exactly know what you already have running with regards to Alexa: does one of the pages mentioning working with Alexa help you out: https://github.com/Aircoookie/WLED/search?q=alexa&type=Wikis ?

Aircoookie, 2020-05-05:
Hi! Wow, that lamp looks absolutely amazing! Great work :)
Controlling an extra relay separately from the RGB lights is not something WLED is designed to do, however you can modify the code to add that functionality.
For that, make sure you can compile WLED from source unmodified first.
Then, change `#define ESPALEXA_MAXDEVICES 1` in line 71 of the `wled.h` file to 2. After that, just follow the API documentation on https://github.com/Aircoookie/Espalexa to add a new EspalexaDevice to the `alexa.cpp` file :)

dnaoci4414, 2020-05-06:
absolutely fantastic I could do it I love your work!!

closed 2020-05-06 by dnaoci4414