wled/WLED #897: i have error compiling for normal IDE example (wifimanager - autoconnect)

aymanokail, 2020-05-05:
i receive this error message when i try to compile knowing that i didn't modify anything from the example in IDE file menu, here is the error:

In file included from C:\Users\Ayman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\Ayman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\Ayman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/Arduino.h:32,

                 from sketch\AutoConnect.ino.cpp:1:

C:\Users\Ayman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                  ^

C:\Users\Ayman\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:32:24: note: in expansion of macro 'PROGMEM'

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                        ^

In file included from C:\Users\Ayman\Documents\Arduino\libraries\WiFiManager\examples\AutoConnect\AutoConnect.ino:5:0:

C:\Users\Ayman\Documents\Arduino\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

exit status 1
Error compiling for board Generic ESP8266 Module.

Aircoookie, 2020-05-05:
Hi!
WLED does not use WiFiManager nor is it supported - maybe you made the issue in the wrong repo?
The functionality provided by WiFiManager is built into WLED :)

aymanokail, 2020-05-05:
hello, i have solved this problem, just removed my old libraries and installed again for the correct version and it worked.
but i have another issue: 1st i was having arduino mega connected with esp8266 module and i had successfully uploaded the sketch by IDE software on the mega and the relays were working properly from the blynk app on mobile, but after some times, the esp8266 lost connectivity and i have to restart the power for the mega to get it back connected again, so i decided to install wifimanager on the esp8266 itself standalone (which was my above post) and after installing the libraries then it works fine, but when i plug the esp8266 back again on the mega and try to connect with blynk app it doesn't work, and on the serial monitor it said:
[1527] ESP is not responding
AT+CIPCLOSE=1
AT+CIPCLOSE=1
AT+CIPSTART=1,"TCP","blynk-cloud.com",80

so i think i did something wrong, can i install a program in the esp8266 then connect it to mega (which has its own program) so i can get use of both, normal program of mega for  my lights synced with blynk app + wifimanager installed on the esp8266?

sorry for the long story but am still new in the IOT and do not know exactly what to do, i have been trying for more than 4 hours without any clue.

Aircoookie, 2020-05-05:
Sorry, it does not appear your problem is related to WLED.
I would recommend asking these people, they might be able to assist you better:
https://github.com/tzapu/WiFiManager
https://github.com/blynkkk/blynk-library
Have a nice day :)

closed 2020-05-05 by Aircoookie