wled/WLED #899: 0.10.0 Hue Sync - JSON Error

DWiskow, 2020-05-06:
I have flashed the pre-built "WLED_0.10.0_ESP8266.bin" onto an WeMos D1 Mini and am getting a consistent "JSON Error" when setting/saving the WLED config to synchronise with a Philips Hue Bridge . . .otherwise everything is working as expected.

The Bridge in question is an old  v1 (round) bridge running SW 01043064 . . . is there a comparability issue with older Philips Hue Bridges and/or can you confirm the 0.10.0 release is working with the Philips Hue Bridge?

Aircoookie, 2020-05-06:
Hi!
I can confirm that 0.10.0 works with my Hue bridge v2. Unfortunately I don't have the round v1 hardware, so I can't say for sure whether it is compatible. Possibly the API was changed as well with the hardware revision?
Is anyone running WLED successfully with a v1 hue bridge?

DWiskow, 2020-05-06:
I have just ordered a Hue Bridge v2 upgrade (£45 complete with a Hue White and Colour Ambience E27 bulb) in the special deal Philips are offering as the v1 goes out of support in July 2020 . . . 

. . . I should have it in the next few days. If the same version of code (0.10.0) works with that, then I will confirm that this is a Hue Bridge v1 issue. If that is the case then this issue could be closed as the v1 will shortly cease to exist.

I will let you know the outcome of my test.

DWiskow, 2020-05-07:
Hue Bridge v2 arrived . . . plugged v2 in and configured it to replace old v1 . . . now syncs without JSON error, and WLED ESP8266 LED strip follows nominated hue Hue light bulb changes as expected.

Please go ahead and close the issue, this is clearly an out of date API in v1 Bridge.

By the way, should my nominated Hue bulb follow changes I make locally to the WLED ESP8266 LED strip? . . . it would seem logical that this function should work in both directions, but it only seems to work from Hue -> WLED?

huggy-d1, 2020-05-08:
Philips was halting API support for v1 bridge.  Here's a quote from the Hue people:

> After April 30, 2020 no software updates will be made available for the Hue Bridge v1 and compatibility with our online services will be terminated at that time. The Hue Bridge v1 can be only controlled locally via the dedicated Philips Hue Bridge v1 app.

DWiskow, 2020-05-08:
@huggy-d1, we get that . . . the question was “is there is difference between the v1 API and the v2 API?” . . . the conclusion, from testing, is there is AND wLED is not compatible with Philips Hue v1 (but it’s not an issue as Philips Hue v1 is disappearing), although it is useful that people are aware their v! Hue will not work with wLED

closed 2020-05-08 by DWiskow