wled/WLED #900: Error 2 NodeMCU

Skyking721, 2020-05-06:
Flashed latest bin release NodeMcu and a few issues I did not see on the D1 mini.  Red Error 2 box on WLED Ap /UI pops up every enter, auto discovery not fining device,  Google mesh network, Mcu locks up on 5g band. (as expected. Cant turn off 5g nor Vlan w/ Google Mesh.) :(

Aircoookie, 2020-05-06:
Hi!
5GHz is not the issue, most every router also accepts connections on 2.4GHz.
Error 2 means the flash layout is off, probably you flashed the `ESP8266_1M.bin` where the correct one for NodeMCU is just `ESP8266.bin` :)

Skyking721, 2020-05-09:
Thanks for reply. I can not disable 5g on Google mesh and therefore when d1 or nodemcu is closer to ap, 5g is assigned by default.   I flashed a D1 with same bin as Node and get same Error on both Node and D1. Strange.

Skyking721, 2020-05-09:
I'll try the other bin on the node. Tx

closed 2020-06-05 by Aircoookie