wled/WLED #902: Software Power Button Preset

jschafermo, 2020-05-07:
A feature request I would like to suggest is a default software power on preset.  For example, the police effect is active when WLED is shut off by the software power button.  Would like a preset, such as a solid effect, that could be set when the software power button is turned on.  Like the preset at boot, but a preset at software power on button as well.  In short a preset that is tied to the software power button.

Just a thought for a feature, but WLED is great already!

Aircoookie, 2020-05-08:
Good idea!

blazoncek, 2023-01-25:
I really don't see a benefit of such function as you can save preset which includes brightness setting and selecting such preset will turn WLED device on and apply any included effect.
If on the other hand you want to preserve state when WLED device looses power you can use Auto Save usermod.

closed 2023-01-25 by blazoncek