wled/WLED #903: Feature request: sync groups

ceeeekay, 2020-05-07:
Hi there, new to WLED and very impressed. I'm installing it around the house for fill lights and had a thought that it would be nice to make groups of WLED clients for separate rooms or areas, so when the sync feature is enabled, colour or effect changes only update lights in a particular group/room. That way you could leave sync enabled and easily update an entire room without affecting other WLED clients in the house.

Any chance of that as a feature, or is there already some way to achieve it? The only way I can think to do it at the moment is subnet each room, which seems like overkill.

Thanks!

fishbone-git, 2020-05-07:
There is an option for udp port number in the sync settings. Only those with the same port number will sync together.  I do this myself to have one grouped set and two separate strings

ceeeekay, 2020-05-07:
Oh true. How did I miss that? Thanks :)

closed 2020-05-07 by ceeeekay