wled/WLED #904: How I save default color

sentinelapr, 2020-05-08:
How can I record the default RGB color, so that when I reboot I don't lose the color settings? I'm new. I put the colors Red Green and Blue and when I turn off the NodeMcu it is placed Orange, No color and no color

raoulteeuwen, 2020-05-08:
You are using segments? Currently, only memory location 16 can store segments. In one of the next release the dev will provide more storage locations and every location will be able to hold segment info. So for now: store your segment setup in location 16, and make the nodeMCU boot up with memory location 16 and you should be good.

sansillusion, 2020-05-08:
I think sentinelapr is saying that after a power off when it boots it loads
default color or stored settings but would like it to remember last state
instead. Wich I too would like a lot since I get random reboots due to heap
corruption on esp32.

sentinelapr, 2020-05-08:
Yes Sansillusion. Sorry for my english.  Its the first time power up.  No configuration.  I choose RGB color, reboot and no remember last setup or save.

sentinelapr, 2020-05-10:
> You are using segments? Currently, only memory location 16 can store segments. In one of the next releases the dev will provide more storage locations and every location will be able to hold segment info. So for now: store your segment setup in location 16, and make the nodeMCU boot up with memory location 16 and you should be good.

No segments, default configuration, firstimer

pille, 2020-05-11:
save the effect settings you want to come up at boot in a slot and configure it to be loaded on boot in LED settings.

closed 2020-09-16 by stale[bot]