wled/WLED #905: Slot 16 cannot be loaded when it saved.

tihoangyeudau, 2020-05-08:
when i save my present to slot 16, it cannot be loaded again. 
every time I click on it I get a notification that slot 16 is empty. Can you fix it?

raoulteeuwen, 2020-05-08:
- What software are you on? One of the latest 0.10.0 bins from [this](https://github.com/Aircoookie/WLED/releases) page? An older version? Self compiled?
- Have you been using WLED before? Has it ever worked for you?
- What exactly do you do in an attempt to store the effect you have running?

Legsmaniac, 2020-05-08:
Wasn't this already confirmed as a bug?  https://github.com/Aircoookie/WLED/issues/891#issuecomment-623364379

Aircoookie, 2020-05-08:
Yes, this is confirmed. Sorry for the visual bug, if you change to a different effect, you should see the effect from slot 16 being applied correctly despite the UI saying otherwise. Will upload a fix soon!

closed 2020-05-08 by Aircoookie