wled/WLED #907: [FR] Link/Sync Segments

Brausmith, 2020-05-10:
Segments are great.
But, I would the ability to link/sync segments together.
e.g.
If I have 3 or more segments, I would like 2 or more of them to have the same effect and with the same timing.

Aircoookie, 2020-05-10:
Hi!
This is currently possible, just the process might be a little un-intuitive, something I could improve in the future. To set multiple segments to the same config:
- Select the segments (tick the checkbox next to them)
- Set the effect you like
- Set speed and intensity sliders (**even if they appear to be correct, often speed and intensity 0 is set for all but the first segment**)
- If segments are still out of sync, switch to Solid mode and back to your desired effect again
- If there are residual LEDs staying on, turn off the power in the web UI and back on again (this can happen when loading from preset slot 16)

Hope it helps :)

Brausmith, 2020-05-11:
Just a quick note, if you do happen to implement a link/sync segments feature in the future, I would like for the 'Reverse direction' option to be excluded from this.
e.g. If I have two segments which are mirrored, then I would also like the option for the effects to remain mirrored.

Aircoookie, 2020-05-11:
Of course, Reverse direction is always a property unique to the Segment and not synced. It's primary purpose is indeed to "mirror" an effect in another Segment which often looks very cool!

Antrit, 2020-05-27:
Hello! I put a couple of problems in the preset cycle after turning off the power, all values ​​are reset and the second question during synchronization is not clearly synchronization sometimes delays or does not switch!

closed 2020-10-02 by stale[bot]