wled/WLED #908: [FR] Segment Offset

Brausmith, 2020-05-10:
Some effects have an obvious start and end point, such as Wipe as an example.
But there are times when I would like the start to be offset by a certain number of LEDs.
I would like a _Start Offset_ field for segments to achieve this.

Aircoookie, 2020-05-10:
Would you want the effect to "wrap around" at the end of the segment? That sort of Start Offset is coming soon, see https://github.com/Aircoookie/WLED/issues/788 :)

Brausmith, 2020-05-10:
That other request looks pretty similar. I too want it for shapes that are round and connect end-to-end.
And a wrap-around option would be great, as I've noticed some effects don't do that.

Brausmith, 2020-05-11:
Just thinking about the start and end of a segment a little further, I can think of two things that you might want to happen.
Wrap - The effect seamlessly transitions from the end of the segment and back to the start.
Bounce - The effect reaches the end, and then bounces (reverses) direction heading back to the start.
So rather than a dedicated 'Wrap' option, maybe have a drop-down to select an 'End Transition'?
I can't think of any additional end transitions, but there might be others to add in the future.

Now coming back to my previous 'Start Offset' request, this is still important, because if I use a Bounce End Transition, then I want it to bounce on a particular LED within the segment.
e.g. A window is outlined on all 4 sides with LEDs, with a physical start point at the top-left corner. But let's say I would rather the Start Offset to be at the top-middle of the window.

closed 2020-09-16 by stale[bot]