wled/WLED #911: Oled nodemcu

aloveday, 2020-05-11:
Hi, does you firmware support pretty oled display output for nodemcu? Aaron

Aircoookie, 2020-05-11:
Hi! It is not officially supported, but there is a usermod to output some status info to an OLED screen: https://github.com/Aircoookie/WLED/tree/master/usermods/ssd1306_i2c_oled_u8g2

OliPassey, 2020-05-15:
I'm getting build errors when trying to use this, this line is causing issue;
IPAddress knownIp;

Do i need an additional library for this?
I've installed U8G2, using a NodeMCU2

Any advice would be appreciated

srg74, 2020-05-15:
You can download firmware.bin from here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/0.10_2005030 It's already with OLED support

OliPassey, 2020-05-16:
Many thanks @srg74

OliPassey, 2020-05-16:
> You can download firmware.bin from here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/0.10_2005030 It's already with OLED support

That firmware went on fine, and WLED is working but the screen does not come on, i'm using the same OLED screen as shown in the wiki of your Wemo shield - looks very cool btw!
Is there a way i can get at a log to see whats up?

srg74, 2020-05-16:
@OliPassey There is no way to turn losing on precompiled firmware. And there is many reasons why your display not working.
1. Different model
2. No pull-up resistors on SDA and SCL
3. Miss wired or dead unit.
If none of above is applicable than you have to recompile firmware.

closed 2020-10-04 by stale[bot]