wled/WLED #912: OTA update, How to tell the differnce between 8266 or esp32

klint76, 2020-05-12:
Hi, i have a bunch of dig-uno's in the room space, is there a way to tell if its a 8266 or esp32, so i can OTA update WLED?

jjohnson2k1, 2020-05-12:
You can query the Json API of the device via a web browser 
http://10.0.0.59/json/info
If you look for the data type "arch" it should show you what you are looking for. 
https://github.com/Aircoookie/WLED/wiki/JSON-API

Example of returned data- 
{"ver":"0.10.0","vid":2005030,"leds":{"count":472,"rgbw":false,"wv":false,"pin":[3],"pwr":1636,"maxpwr":50000,"maxseg":10,"seglock":false},"str":false,"name":"WLED","udpport":60606,"live":false,"lm":"","lip":"","fxcount":103,"palcount":52,"wifi":,"rssi":-30,"signal":100,"channel":1},
"arch":"esp8266","core":"2_6_3","lwip":1,"freeheap":6856,"uptime":19835,"opt":119,"brand":"WLED","product":"FOSS","mac":""}

klint76, 2020-05-12:
Awesome, thanks for your time, and speen in reply, this info is very useful.

Thanks again

Klint

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

huggy-d1, 2020-05-12:
You could rename your devices as you remotely discover esp8266/esp32.  Let's say you mostly have ESP8266's running WLED.  Leave the names of those alone.  The esp32's, rename those to "existingname_32" or "existingname-32", whichever works for you.
Then, next OTA update time, you will know just from the device's name without having to interrogate the JSON data, and inspecting the resulted results first.

sansillusion, 2020-05-13:
Use this after the ip of your wled ...
/json/info

Eg : http://192.168.1.140/json/info

closed 2020-09-17 by stale[bot]