wled/WLED #913: APA102C don't get white color

MarioPeters, 2020-05-12:
I use a nodemcu 8266 with APA102C and everything works fine, but I don't get a white color.
If i'm choosing white, I am getting red one.
I've found out, that if I reduce the brightness to minimum I will get the white color.

I've chose GRB. 

Does WLED only works with APA102 instead of APA102C, or what I am doing wrong?

Aircoookie, 2020-05-12:
Hi, this sounds like an insufficient power issue. APA102C should work fine, how many do you have and what is your power supply?

MarioPeters, 2020-05-12:
I've testet with 26 LEDs and 2A, but I will try another power supply.
Another Question:
Is it possible to connect 2 LED stripes at one ESP or have I to use 2 ESP ?

huggy-d1, 2020-05-12:
If you are ok with both LED strips having exactly the same effects at all times, you can use the data pin from your ESP to drive both LED strips.  However, the strips need to be pretty close to the ESP, or use a level shifter to increase the voltage and also make it easier to drive both LED strips.

MarioPeters, 2020-05-12:
The power supply was the problem

MarioPeters, 2020-05-12:
I used 5m telephone cable from the stripes to the ESP. No Problems.

closed 2020-05-13 by Aircoookie