wled/WLED #916: Add new Confetti effect

nordeep, 2020-05-15:
Could you, please, add new Confetti effect based on https://github.com/atuline/FastLED-Demos/blob/master/confetti/confetti.ino

Aircoookie, 2020-05-15:
Hi, check out Twinklecat, that effect should be very similar! The only thing it doesn't yet have are the automatic color changes, though you can control the colors you'd like by setting a palette. You get maximum "Confetti" with the intensity slider set to 255. I'll look into adding the Confetti effect you linked!

nordeep, 2020-05-15:
Thank You! The similar, but not the same :).
In my local version of confetti effect I additional change fadetToBlackBy speed like a:
fadeToBlackBy(leds, NUM_LEDS, beatsin16( 6, 2, 50 )) - it gives "the breathe" effect
and
Density of confetti - count of leds to be colored speckles in one pass. Good effect if density=NUM_LEDS/20:

```

settings.ftb_speed = beatsin16( 6, 2, 50 );
settings.confetti_dens = settings.num_leds/20;

void confetti() {
  // random colored speckles that blink in and fade smoothly
  fadeToBlackBy(leds, settings.num_leds, settings.ftb_speed);
  for (int x=0; x<settings.confetti_dens; x++) {
    int pos = random16(settings.num_leds);
    leds[pos] += CHSV(gHue + random8(64), 200, settings.effect_brightness);
  }
}
```

closed 2020-09-19 by stale[bot]