wled/WLED #919: Saving Segments to Preset 16 not working.

FalkPe, 2020-05-16:
Hello,
I use WLED 0.10.0 (build 2005030) on my NodeMCU v3. I have connected about 5 meters of WS2801 in 3 Segments. 
After power loss my segments are lost so I read that I can save my config to Preset 16.

But this will not work. When I force to save to 16, I get the message, Saving the config to slot 16. But this will remain not highlighted. 1-15 works for me, and after power loss I can load them. 

Just 16 won't work. Is there maybe a problem with the release for WS2801 firmware?

Thank you

Aircoookie, 2020-05-16:
Hi! It is in fact working, just set 16 as your boot preset in LED settings. There is a bug in 0.10.0 where Preset 16 says it is empty even though it is not: #891 
It is already resolved in source, sorry for the issue!

closed 2020-05-16 by Aircoookie

FalkPe, 2020-05-16:
Wunderfull, sorry for my not so good research 👍

FoxExe, 2020-09-07:
I think need add this info to Readme file, or just fix it - save segments info independed of saving slot. Or save in each slot for different configurations (different animation per segment)