wled/WLED #920: strange IR readings

brainbot-github, 2020-05-17:
first, thank you Aircookie and all contributors for this amazing project :D 

I tried to use a salvaged IR sensor from a random chinese LED driver which also used a generic 24-key IR remote. I kind of figured out which sensor it is ( probably TSOP38238) and wanted to use an old TV remote. I get readings, but they are pretty much random. About every tenth command or so has the same hex value, but everything in between seems to be random. Any idea what could cause that issue or what I'm doing wrong? Thanks in advance

Aircoookie, 2020-05-18:
Strange... Maybe you could try adding a 10k pullup resistor between the signal and 3.3v lines. Or try a KY-022 sensor module to verify that it is not an issue with your sensor. What brand is the TV remote? It might be that the IR library is not compiled to recognize the IR codes of that manufacturer. In that case we could compile a custom version that supports it!

brainbot-github, 2020-05-18:
Thanks for your replay. The second attempt was with: 10k Pullup between 3v3 and data and, as mentioned in the datasheet of the sensor, a series resistor in the 3v3 line (after the pullup connection) and a capacitor, but to no avail. Today, I bought a new receiver (TSOP31238, my electronics dealer had only this one for the 38kHz range) and it's working much better. I can't tell about the brand of the remote, most of the printing is worn out, except for some useless "Designed by [ unreadable handwriting in plastic] ". The received codes are much more constant now. As ordered in the "How to custom IR", I only briefly press the button, and now at least every second code from the same button is identical (e.g. IR recv 0x47997125   ;   IR recv 0xA4647371). Now, I'm entering some of the codes to see how it will behave. Also, with the #1 remote (white, 24 button, 12 colors), the codes are the same for each button now. So my salvaged sensor was also causing issues.

closed 2020-09-23 by stale[bot]