wled/WLED #922: TPM2.NET

Legsmaniac, 2020-05-17:
Am I correct in thinking that you cannot use TPM2.NET with WLED in it's current form?  I've tried playing around with the Sync Interfaces settings page, setting the port to 65506 in both WLED Broadcast and Realtime without any success.

So if it's not possible, could it be added please?  I have used [this](https://gist.github.com/delhoume/2688aa50d798dbb94c9adf8e77f0454b) code stand-alone and it works fabulously.  

I have used both the above code and WLED on my home-made matrix of 2040 LEDS and both work fine with no problems.  Of course, I had to modify the WLED source to allow more than 1500 LEDS and I also turned off the annoying (annoying  to me lol) warning when selecting above 8.5amps (I set 30 amps for my matrix.)  I've read elsewhere that Matrix inclusion is planned for the future so I feel that 1500 LEDS is rather limiting with todays tech and code.  Yes, I understood the warning was because of stability and lag issues but I had none of either on my 60X34 matrix.  This is using WS2812B 60 LEDs per metre LED strips, ESP8266 NodeMCU 4M and the latest WLED Master v10 code.   

Anyway, I digress.....
TPM2.NET?  😃

Aircoookie, 2020-05-18:
You are correct, TPM2.NET is currently not supported by WLED. Since the protocol is REALLY simple though, I believe I can get it added quite quickly!

Legsmaniac, 2020-05-18:
That would be fantastic, thank you.

Aircoookie, 2020-05-18:
And support is in! You will need to change the WLED broadcast port to 65506 for it to work. I don't know if it is stable with 2000 LEDs (although I modified the code you linked to support multiple packets/blocks), if the performance is worse than with E1.31/Art-Net we could move the implementation to async.

Legsmaniac, 2020-05-18:
OK, I finally found the time tonight to test this.  The good news is - it works perfectly on my small test panel of 16X16 (256) LEDS.  The bad news is, it didn't work at all on my 60X34 (2040) LEDS matrix.  (Whereas the code I linked to installed as stand-alone does work)  There was no change in the matrix at all, it just continued to play one of the WLED effects.  If I changed the Chan/Block in Jinx! from 6120 to something lower such as 765 (8 blocks), then I got an output but it was pretty much a broken mess.  6120 channels, 1 block works fine with the original code.

Oh, and I also had to modify the source in order to get it to let me set up 2040 LEDS before I could start, it told me 1500 max., preferably 1000 or less.  

Happy to test the next modification if and when you get around to it but no rush as I imagine you must have a lot on.  Since the stand-alone code works, I imagine it must be able to be done.  You know best.  😉

Legsmaniac, 2020-05-19:
> if the performance is worse than with E1.31/Art-Net.....

BTW, I did try E1.31, works well but isn't it limited to 512 channels?  Or am I missing something?  This limit was the reason for my request for TPM2.

Aircoookie, 2020-05-19:
Interesting, I thought that the maximum UDP size was 1472 bytes, so WLED will refuse any packets above this size. With the stream read from the code you linked it _should_ work however, so I'll try that! (reading directly from UDP packet rather than creating a buffer) I believe I know the reason for the mess when sending the data in multiple blocks, which is packets arriving out of order. Will try and fix that soon :)

About E1.31/Art-Net, it is limited to 510 channels (170 LEDs) per universe. However, WLED can receive 9 subsequent universes which bumps up the support to 1530, about the same as the default LED count limit. You can increase `#define E131_MAX_UNIVERSE_COUNT` in const.h to 12 universes, that will exactly cover your 2040 LEDs :) Seems like Jinx only supports one universe though :/ I swear I remember it supporting multiple...

Legsmaniac, 2020-05-19:
> I thought that the maximum UDP size was 1472 bytes

That was my understanding too, so I under the impression I was going to have to "split the payload" as per instructions in the code.  However, I typed into Google "What is the maximum UDP size?" and got the answer......

65,535 bytes
The field size sets a theoretical limit of 65,535 bytes (8 byte header + 65,527 bytes of data) for a UDP datagram. However the actual limit for the data length, which is imposed by the underlying IPv4 protocol, is 65,507 bytes (65,535 − 8 byte UDP header − 20 byte IP header).

Re E1.31 - It's possible I could use Parallel Inputs on my matrix but because of the fact that it is created in strips of 60 leds per row, I would be looking at utilising two rows (120 leds) per universe rather than 170 leds, otherwise I'd be looking at having to cut my strips up and injecting data feeds in the middle of the matrix board. And to use 2 strips/120 leds means 17 Universes for 2040 leds!   Sod that!  LOL   

So I would really rather use TPM2 and preferably a single data feed if I can though I have experimented with 7 parallels (one per 5 metre reel or 300 leds, 5 strips in my case) using an ESP8266 with success.  (Yes, I ignore the rules and try anything once! LOL)  But all Parallels are doing is saving processing time, ideal if I was going to use my matrix as a TV but that is not the case so one data feed is quite sufficient for my needs.  🙂

closed 2020-10-12 by Aircoookie

ayandeepd1, 2022-08-03:
> So if it's not possible, could it be added please? I have used [this](https://gist.github.com/delhoume/2688aa50d798dbb94c9adf8e77f0454b) code stand-alone and it works fabulously.

Did this code work out of the box? there is a mention of using numbered packets in this code if leds exceed 484(21x21)? however you say you have 60x34 leds and they worked fine.  i have to run a 90x30 broken equally into 3 panels each having its esp8266 and respective source outputs. is this supposed to work or am i to use packet numbers and change the channels/block in output source settings?