wled/WLED #923: Preset cycle - time to cycle

TymmyD, 2020-05-18:
Hi. I've read about the preset cycle turning the lights back on and looking forward to a fix or hearing of a temp fix. I can not seem to put the amount of seconds in and make them stick. I'll put in 300 seconds. Tick the box and it goes back to 54.6 or something. 60 seconds works fine. But would be nice to have it change every 10 mins or every hour. Unless it's just me? 
Great work on WLED!

renearts, 2020-05-18:
I have encountered this issue a few days ago as well and have been looking into it. Preset cycle times above 65536 milliseconds are currently not supported. Yes, the time is internally defined in milliseconds which is fine in itself, but I was looking for a longer cycle time as well. Say at least 10 minutes up to an hour or more. I use this project for outdoor LED strips lighting up my garden. Changing presets every minute is quite short in my opinion as it feels a bit restless. 

The main culprit for this is that the presetCycleTime is defined as a 16 bit unsigned integer, hence the maximum value of 65536 (2^16). Changing this to a uint32_t would IMHO be sufficient as it would allow cycle times of almost 50 days. 
`WLED_GLOBAL uint16_t presetCycleTime _INIT(1250);
`
Did a quick try in changing uint16_t into uint32_t in line [418](https://github.com/Aircoookie/WLED/blob/579f3558521defd37ea522e7e60a73fb588b8a97/wled00/wled.h#L418) in wled.h does not directly yield good results. We'll have to do a bit more I guess. But at least good to see that more people are running into this.

Aircoookie, 2020-05-18:
This is correct, 65.5 seconds is currently the maximum duration per preset.
The API for this is already in tenths of seconds, so I will update the internal variable to be based on that as well. That will yield a maximum duration of over 100 minutes.

renearts, 2020-05-18:
I might have found the issue and a solution; I also needed to expand the save to eeprom to have the correct width; it only saved 2 bytes which needs to be 4. I'll try to test it tonight or tomorrow and I might be able to send a pull request.

Aircoookie, 2020-05-18:
Having 32 bits could be viable, but as you said I didn't want to change the EEPROM layout. The ESP-side code now supports up to 6553 seconds, I just need to adjust the HTML input field max to make it accessible from the UI soon!

TymmyD, 2020-05-18:
Thank you for your help and support.

closed 2020-09-04 by Aircoookie