wled/WLED #924: mDNS not connecting

Tanner8021, 2020-05-18:
I've tried about 5 times to connect my NodeMCU to my Wi-Fi, but evertime I type in my mDNS address, I get "wled.local’s server IP address could not be found." 

I assume that the NodeMCU is connecting to my Wi-Fi as it is no longer an open network.

My network is open, so I leave the password blank, and my Wi-Fi has 2.4 and 5G (I've tried both)

Thanks for your help!

Aircoookie, 2020-05-18:
Hi!
2.4G wifi is correct, an ESP cannot connect to 5GHz networks.
Have you changed the mdns in wifi settings? By default it is not `wled.local`, but something like `wled-20fd3a.local`.
Check in your router configuration for the IP of the NodeMCU to see if it connected successfully. You can also try to discover it in the WLED app. If it is found mDNS works :)
Lastly, some few routers seem to block mDNS packets. You could check if there is a setting for that in your router config.
I hope you can get it to work soon! :)

KM-LED, 2020-06-29:
Hello
First sorry for using google translator.

I have the same problem, both in nodemecu and in wemos d1 mini.

I fixed it as follows.
I enter the web of configuration, complete all inclusive dns.
when i want to enter it gives me the error that it can't find the address ..
you find use the advance port scanner application, there i set the ip address assigned to me.
In my modules the manufacturer is espressif inc.
Then I look at the ip that was assigned to me 192.168.30.301 I copy it into Chrome and ready ...
If not another possibility is to add the ip address yourself ...

I hope it works ... I don't speak English ...

Greetings to all..

sansillusion, 2020-06-29:
On windows machines I need to install Apple's bonjour service to get mdns
to work.

Tanner8021, 2020-07-02:
It works with the port scanner and the IP! Thank you so much @KM-LED

ThePowertoaster, 2020-09-24:
@sansillusion hit it right on the head.  installing apple bonjour print service immediately made mDNS work on my machine,

closed 2021-01-31 by stale[bot]