wled/WLED #925: Of no major importance but...  a small typo?

Legsmaniac, 2020-05-18:
in wled00 is says in the comment at the top.......

 * Usermod compatibility: Existing wled06_usermod.ino mods should continue to work. Delete **usermod.cpp**.
 * New usermods should use **usermod.cpp** instead.

Contradiction?  Shouldn't that be Delete usermod.ino?  😕🤔

Aircoookie, 2020-05-19:
No, this is intentional :)
If you want to use an old .ino usermod, you'll need to delete the .cpp or it will not compile because the functions are defined twice. Otherwise just replace the .cpp with your usermod.

closed 2020-05-19 by Aircoookie

Legsmaniac, 2020-05-19:
Ah!  That makes more sense now you've explained it like that.  Sorry.  Or maybe it was just me being thick as usual.  😂

Thanks