wled/WLED #926: ESP32 and APA102?

pierrebuet, 2020-05-19:
First of all, this is more a question than an issue. I'm using a binary of WLED for ESP8266 and APA102 and everything works very well. I wanted to give a try to an ESP32 to drive an APA102 strip, but I'm having trouble to understand which binary I should pick up. I find binaries for ESP32 but none for ESP32 AND APA102 (and the pins needed for). Am I perhaps missing something?

huggy-d1, 2020-05-21:
What pins are you trying to use?  If you are using CLK = GPIO1, DATA = GPIO2, Quindor has a working binary posted in his Discord, which you can find a link to on https://quinled.info/ for the QuinLED DigUno board.

huggy-d1, 2020-05-21:
From Quindor's info site (linked above), for the ESP32, he states (for the DigUno & ESP32): 

> Using the ESP32 version the GPIO2 pin is connected to the onboard LED, 
> the pin on the QuinLED-Dig-Uno is connected to GPIO16 instead!
> To make an ESP32 module work with APA102 for instance I needed to use
>   #define CLKPIN 3
>   #define DATAPIN 16

pierrebuet, 2020-05-21:
@huggy-d1 : I'm actually free to use the pins I want. I soldered then with CLK = GPIO0 and DATA = GPIO2, but this is something I can change easily. Unfortunately, I was not able tp find a binary to flash following the link you gave me. Do you perhaps have a little bit more information for me to find it?

closed 2020-09-26 by stale[bot]

huggy-d1, 2020-09-28:
@pierrebuet Did you find a bin for your project?  If you check the released firmware on WLED, they are now building and posting more bin files.  Maybe one of them is exactly what you need?

r100gs, 2020-10-03:
I would like to use ESP32 and APA102, but there is still no bin, or is this wrong?

Best regards,
Stefan

Elkropac, 2020-10-13:
Hi, i have diguno which i know works with WS2812B. Now some SK9822 arrived i want to try.
I compiled wled from git with 
```
[env:custom32_LEDPIN_16]
board = esp32dev
platform = espressif32@1.12.4
build_flags = ${common.build_flags_esp32} -D LEDPIN=16 -D USE_APA102 -D CLKPIN=3
lib_ignore =
  ESPAsyncTCP
  ESPAsyncUDP
```

but leds does not react. Is it proper way to compile firmware like this? How to debug?