wled/WLED #927: Esp32 max led?

Iv711, 2020-05-19:
Hey. I see on the wiki a mention of the recommended 750 LEDs. Is it unified for both esp32 and esp8266?
If so, how much is it possible to use 900-1100 LEDs on esp32?
Thx!

Aircoookie, 2020-05-19:
Hi, it is entirely possible to run about 1k of LEDs on ESP32 - with ESP8266 one can run into memory issues at high LED amounts, that is not an issue with ESP32.
The main limitation is the LED refresh rate, WLED uses 42fps (achievable with 750 LEDs). With 1024 LEDs, you'll be getting 30fps. For slow effects that is enough, but you might get some slight stutters if you run very fast chase effects :)

Legsmaniac, 2020-05-19:
> I see on the wiki a mention of the recommended 750 LEDs.

I've come across this before, on at least one other site and quite possibly on one of the many YouTube videos.  The reason it's possible to use more LEDs is the FastLED and NeoPixel libraries apparently, the software being designed to handle the memory, or so I was told.  I've used 2040 LEDs on a 60X34 Matrix I made using a NodeMCU ESP8266 and WLED with no problems whatsoever.

s1nbad, 2020-05-19:
I currently have 630 pixels on an ESP32 which has been faultless ...but
I am experiencing problems running E1.31 delivered by LedFX (even at 10 fps)

before troubleshooting further (DMX) do the pixel limits change when using E1.31

s1nbad, 2020-05-19:
> I currently have 630 pixels on an ESP32 which has been faultless ...but
> I am experiencing problems running E1.31 delivered by LedFX (even at 10 fps)
> 
> before troubleshooting further (DMX) do the pixel limits change when using E1.31

Started troubleshooting and i can answer my own question  - yes they do, see
https://github.com/Aircoookie/WLED/wiki/E1.31-DMX

huggy-d1, 2020-05-20:
My understanding is 512 is the pixel limit for reliable E1.31

pille, 2020-05-22:
> My understanding is 512 is the pixel limit for reliable E1.31

it should be 510, if the reliability comes from a single universe. multiple universes are supported, though.

DutchmanNL, 2020-06-08:
> I currently have 630 pixels on an ESP32 which has been faultless ...but
> I am experiencing problems running E1.31 delivered by LedFX (even at 10 fps)
> 
> before troubleshooting further (DMX) do the pixel limits change when using E1.31

i had the same using Hyperion with E1.31 the colors were incorrect (used 462 WS2812B)
Changed it to UDPRaw and now all is OK

h9h8h7, 2022-11-30:
hello! can you send to me a doc of main steps to create beautiful diy mods?