wled/WLED #928: Update interval for static mode

SamPUG, 2020-05-19:
I had some issues using overlays due to the default update interval for the static mode.

The `handleOverlayDraw` function, which updates the overlay, is called  based on the update interval. This is normally 500ms for the static show as shown below; much longer than most effects.
```
/*
 * No blinking. Just plain old static light.
 */
uint16_t WS2812FX::mode_static(void) {
  fill(SEGCOLOR(0));
  return (SEGMENT.getOption(7)) ? FRAMETIME : 500; //update faster if in transition
}
```
Personally I think the 500 ms delay between updates is a little long and could be limiting for overlays. I have been playing around with a more advanced clock overlay and it has caused a bit of issue. 

Personally I think `mode_static` should always just return `FRAMETIME` to make the behavior more simple and  make the strips update rate the same as the rest of the modes.

Aircoookie, 2020-05-19:
Yes, I'd rather have it on `FRAMETIME` like the other effect as well, and you can absolutely do that if you rely on a higher refresh rate for a custom overlay. The problem and reason for the 2fps refresh rate is that many installations are not well connected, there are cases where the 3.3v data signal is just enough to drive the LEDs most of the time, but they may flash up brightly every few minutes because of faulty transmission. This low refresh rate limits that very effectively. Of course this is just a workaround, but not everyone is prepared for the actual solution, which is to use a levelshifter to get a stable signal.

SamPUG, 2020-05-19:
Thanks for the explanation, that makes sense. I have done that for now but I was trying to keep things as close to the master as possible.  

Thank you for your hard work on this excellent project!

closed 2020-05-22 by Aircoookie