wled/WLED #929: Will not compile with warnings severe enough to be treated as errors

marcmerlin, 2020-05-20:
In arduino preferences, I have "warnings: all" because it's the only way to find so many programming errors. In a few libraries, it gives warnings that can be cleaned up.
With NeoPixelBus, the warnings are severe enough that the compiler treats them as errors and refuses to build.
I can indeed turn them all off and then it builds, but basically I'm building code full of pretty bad warnings.

I'm using arduino 1.8.9 building for ESP32 using neopixelbus 2.5.7.
/home/merlin/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H (snip) I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/fb_gfx -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10809 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=4 

```
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:79:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:88:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:103:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:112:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:125:119: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     __attribute__((noinline)) static const void ICACHE_RAM_ATTR send_pixels(uint8_t* pixels, uint8_t* end, uint8_t pin)
                                                                                                                       ^
In file included from /tmp/arduino_build_684949/sketch/FX.h:30:0,
                 from /tmp/arduino_build_684949/sketch/fcn_declare.h:79,
                 from /tmp/arduino_build_684949/sketch/wled.h:88,
                 from /tmp/arduino_build_684949/sketch/xml.cpp:1:
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In constructor 'NeoPixelWrapper::NeoPixelWrapper()':
NpbWrapper.h:326:53: error: 'NeoPixelWrapper::_pGrbw' will be initialized after [-Werror=reorder]
   NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;
                                                     ^
NpbWrapper.h:322:16: error: 'NeoPixelType NeoPixelWrapper::_type' [-Werror=reorder]
   NeoPixelType _type;
                ^
NpbWrapper.h:157:3: error: when initialized here [-Werror=reorder]
   NeoPixelWrapper() :
   ^
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::Begin(NeoPixelType, uint16_t)':
NpbWrapper.h:176:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type)
            ^
NpbWrapper.h:176:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::Show()':
NpbWrapper.h:266:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type)
            ^
In file included from /home/merlin/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h:96:0,
                 from /home/merlin/Arduino/libraries/NeoPixelBus/src/NeoPixelBrightnessBus.h:29,
                 from /tmp/arduino_build_684949/sketch/NpbWrapper.h:144,
                 from /tmp/arduino_build_684949/sketch/FX.h:30,
                 from /tmp/arduino_build_684949/sketch/fcn_declare.h:79,
                 from /tmp/arduino_build_684949/sketch/wled.h:88,
                 from /tmp/arduino_build_684949/sketch/wled_serial.cpp:1:
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:79:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:88:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:103:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:112:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:125:119: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     __attribute__((noinline)) static const void ICACHE_RAM_ATTR send_pixels(uint8_t* pixels, uint8_t* end, uint8_t pin)
                                                                                                                       ^
NpbWrapper.h:266:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h:265:10: warning: unused variable 'b' [-Wunused-variable]
[... 548 lines cut ...]
NpbWrapper.h:313:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::cleanup()':
NpbWrapper.h:330:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type) {
            ^
NpbWrapper.h:330:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
'NeoPixelWrapper::_pGrbw' will be initialized after [-Werror=reorder]
```

huggy-d1, 2020-05-20:
Are you using VS Code & PlatformIO IDE to help automate downloading of libraries and handle dependencies?

marcmerlin, 2020-05-20:
@huggy-d1 no I'm not using PlatformIO, just the arduino layer.
I installed the dependencies myself, and got the whole thing to compile, but I had to turn off all warnings for the code to actually build.
It did crash when I had DMX on (I didn't need it, so I turned it off), and I got a baisc binary working now.
That said, all those bad warnings in the neopixelbus interface, ought to be fixed, it's not safe to work with code when all compiler warnings are turned off.

huggy-d1, 2020-05-20:
If you are using VS Code without platformIO handling all the library dependencies, you are likely to get all kinds of warnings.

WLED does NOT use the latest version of all libraries.  It uses what works and if/when a good reason to change versions happens, then that library version gets updated in the build process used by platformIO.

I say this because I used platformIO to build WLED and received zero (0) warnings.  And, I do not know much at all about Arduino or PlatformIO.  It was my first successful build and it had no errors or warnings.  Made me day!

marcmerlin, 2020-05-20:
@huggy-d1 aaaah, I see what you're saying, neopixelbus was maybe updated in some incompatible way, and now the up to date git version is throwing all those errors when you build against it?
Are you sure all your error warnings are turned on in platformio? If I build with the arduino env with all warnings turned off, then it builds and works.

I'm defintitely seeing the same problem when building with the bare arduino environment, which arguably is what most users use and are comfortable with.  Have you tried to turn on all your warnings to maxmium level and use the lastest avr-gcc (which likely is more picky about code that has potential issues)?

huggy-d1, 2020-05-20:
I used the default settings for PlatformIO IDE installed to VS Code.  I do not know how generous or how particular the default settings are in comparison to listing every possible warning, like you misspelled a word in a comment.

huggy-d1, 2020-05-20:
On a related note to your suggestion about the arduino dev system.  I tried installing and configuring everything to work (version wise according to the WLED wiki build instructions). In fact, I tried multiple times and I could never get a build to succeed because I felt like I was recursively installing dependent libraries, and then having to figure out which version of the dependent library was compatible with the required library version that depended on it.

I tried it once in VS Code / Platform IO and it just worked.  I'm not that smart.  The software is just that good.

huggy-d1, 2020-05-21:
I will say this from my very brief interaction with both environments.  Once you go to VS Code for Arduino environment development, you won't want to go back.  It will feel so, umm, antiquated and clunky by comparison.  

VS Code/PlatformIO IDE (for Arduino/ESP8266/ESP32/ and so much more) with live Intellisense and auto-complete, automatic library dependency handling, and so much more.

Or, I could have spent even more time stubbornly using a functional, yet more manual, frustrating and _error-prone_ dev tool.

Aircoookie, 2020-05-22:
Hi, thanks for reporting your findings!
It seems like most of the issues are in `NpbWrapper.h`, that file will be completely replaced in v0.12.0 when multi pin output support will be added. Also one of the enum values not being respected in the switch isn't really severe enough to error. It is designed to and will just do nothing in that case.

Although `error: no return statement in function returning non-void` in xml.cpp is something I really should look into.

closed 2020-09-26 by stale[bot]