wled/WLED #930: Alexa problem

colder1989, 2020-05-20:
i have a problem with alexa...when i try to controll my strip with alexa, the Nodemcu go always offline... i need to reboot manual... if i control the led from my browser and with the app works good...
i have compileted by myself but the only change is uncommented analog led and change UDP port from "Sync Settings"

huggy-d1, 2020-05-20:
Have you disabled SYNC for all things that can sync up?

colder1989, 2020-05-20:
[image]
I've got everything off, except alexa.

huggy-d1, 2020-05-21:
I am by no means an expert in configuring WLED (meaning I can be 100% wrong), but.. here's what I see enabled that relates to outside devices:

1. Broadcasting brightness, color, effects all enabled
2. Send notifications for direct change & button press enabled
3. Receive UDP realtime enabled

On the main WLED screen, is SYNC configured to ON for either send or receive?

colder1989, 2020-05-21:
Ty man...now works...if i enable "send alexa notifications", the device stuck..... what is that option for?

huggy-d1, 2020-05-21:
I don't know, but guess it relates to updating Alexa with current values for dim level, color, and on/off status.

d3iu, 2020-06-20:
i can't even get alexa to discover the device, all is set, including a name, alexa option is on... nothing works.. disabled everything on sync... literaly nothing works.. wled is running on a nodemcu, did not have this problem with another device running wled version 9

huggy-d1, 2020-07-29:
@d3iu Did you ever get Alexa to discover your WLED install?

drschnalli, 2020-11-02:
I'm trying to connect it to alexa, too. But it does not discover the device. Alexa Sync is turned on.

closed 2021-06-04 by stale[bot]