wled/WLED #931: Wemos D1 Mini - Alexa Problem

Bihel29, 2020-05-22:
Hello, first sorry for my bad english.

I recently installed WLED on a Wemos D1 Mini to control my Nanoleafs replica equipped with LED WS2812B.

I followed this guide: https://tynick.com/blog/11-03-2019/getting-started-with-wled-on-esp8266/ to install WLED_0.10.0_ESP8266.bin.

Everything works perfectly with the Android app.

I recently purchased an Amazon Fire Stick and would like to use Alexa with my WEMOS.

But unable to find my Wemos with the "Discover devices" function, Alexa tells me that no device has been found.

So I installed the Alexa application on my Android but same problem.

Lamps => Philips Hue V1 => No device found.
Lamps => Other => No devices found.

WLED configure by default, Alexa is activated in the "SYNC Interfaces" tab

Did I miss something ??

Thank you in advance for your help.

Aircoookie, 2020-05-22:
Hi, sorry for the issue!
Unfortunately the Alexa emulation is only supported with a physical echo device (which does not include the Alexa app and - counterintuitively - not the Fire TV stick). You'll need an Echo, Echo Dot, Echo Show or similar for it to work unfortunately. Once you have that device on your network, control should also work via phone or the Fire TV stick.

Bihel29, 2020-05-22:
Oh ok, now I understand why it doesn't work

No library or solution exist to work with a Fire Stick or the Android application?

Thank you for your response and thank you for WLED.

Aircoookie, 2020-05-22:
Not to my knowing. The control works by emulating the local API of a Philips hue bridge. The way Amazon wants you to do it is via a vendor cloud service - something i neither have the resources nor will to set up. Maybe someday Amazon will release a standardized local API - but I don't know if they have any motivation to do so...

Bihel29, 2020-05-24:
Ok , thanks for your answer

closed 2020-05-24 by Bihel29

derekpurdy, 2020-08-28:
@Aircoookie Out of curiosity should an Ecobee work as a Physical Echo Device?

Aircoookie, 2020-09-02:
@derekpurdy I don't think so, I believe 3rd party devices have a more basic version of the voice control API. For example it also doesn't work with Sonos speakers (unless an Echo is also in the network). But if it works definitely let me know!