wled/WLED #932: Dim in Effect and Dim out Effect?

FalkPe, 2020-05-22:
Is it possible to add the called effects?

I use Wled for my stairs, this would be awesome.

Thank u for your nice product

Aircoookie, 2020-05-22:
Hi, I'm glad that you like WLED :)
simple dimming in and out is done by just setting your preferable transition duration and then adjusting the power, brightness or color, which will smoothly fade.
If you want the stairs to turn on one after another, set the Grouping in the Segments tab to your number of LEDs per step and then try the Wipe mode with the intensity slider set to max.
https://github.com/Aircoookie/WLED/tree/master/usermods/stairway_wipe_basic might also be interesting.

FalkPe, 2020-05-22:
Wipe is a cool effect. But only one rotation would be great. 

It seems that I must recompile your code for the stairway wipe basic to add this effect?

Thanks for your help

FalkPe, 2020-05-24:
Hey @Aircoookie ,
I realized it with my Homeautomation.
If you want you can view it here:

https://youtu.be/_Je2XIdSHFw

Aircoookie, 2020-05-24:
Looks cool! Great project, I wish I could do it for my staircase at well but bothg the steps and handrail are kinda unsuitable for attaching LEDs :D

closed 2020-05-24 by Aircoookie