wled/WLED #935: Relay on GPIO12 with RLYMDE=0

Pluto1010, 2020-05-23:
Hello!

@Aircoookie Thank you for this awesome software!

I've discovered recently that whenever I connect a [relay](https://www.ebay.com/i/181026509345) that powers on on LOW (RLYMDE=0) to the default GPIO12, the board ([NodeMCU-32S](https://www.ebay.com/i/123401670127)) is no longer able to boot when power supply (via USB) is turned on. It hangs until I remove the relay for a short time. After boot I can reconnect it and it works fine as expected. 

Due to this behaviour I've decided to use GPIO27 to connect the relay. I use it to power off the bigger power supply when unused. Its driving 340 LEDs.

Maybe we should change the default to another PIN. I don't know if there is a good reason that GPIO12 was chosen. For me it works better on GPIO27 and this pin does not seem to be used for something else right now.

Aircoookie, 2020-05-23:
Hi, glad you like WLED!

Indeed it seems like GPIO12 needs to be LOW on boot to select the proper flash voltage: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins
Your relay is likely pulling the pin up, preventing boot.

GPIO12 was chosen as the default pin as this is the relay pin on (most) Sonoff units. With development boards, you can easily choose a different pin but with existing hardware, less so.
I've got good news however as I plan to make the pin runtime configurable in the near future :)

Pluto1010, 2020-05-23:
That would make it easier.

I think I have just found the issue where this topic has started: #128

closed 2020-05-23 by Pluto1010