wled/WLED #936: Feature request, ESP32 version with switch pin change from 0 to 2

Bleepbloopblahp, 2020-05-23:
Hi, I'd like to request a custom binary for ESP32 with the switch/button pin changed.

I bought a few ESP32 boards to use with WLED but neglected to notice they do not have an output for GPIO 0 on the board.

If there is any way to make this sort of change an option in the config inside WLED I think it could future proof it from these sort of issues.

bjornbergenheim, 2020-05-25:
This can be changed in NpbWrapper.h
(LEDPIN and BTNPIN)

```
//this code is a modified version of https://github.com/Makuna/NeoPixelBus/issues/103
#ifndef NpbWrapper_h
#define NpbWrapper_h

//PIN CONFIGURATION
#ifndef LEDPIN
#define LEDPIN 2  //strip pin. Any for ESP32, gpio2 or 3 is recommended for ESP8266 (gpio2/3 are labeled D4/RX on NodeMCU and Wemos)
#endif
//#define USE_APA102  // Uncomment for using APA102 LEDs.
//#define USE_WS2801  // Uncomment for using WS2801 LEDs (make sure you have NeoPixelBus v2.5.6 or newer)
//#define USE_LPD8806 // Uncomment for using LPD8806
//#define USE_TM1814  // Uncomment for using TM1814 LEDs (make sure you have NeoPixelBus v2.5.7 or newer)
//#define USE_P9813   // Uncomment for using P9813 LEDs (make sure you have NeoPixelBus v2.5.8 or newer)
//#define WLED_USE_ANALOG_LEDS //Uncomment for using "dumb" PWM controlled LEDs (see pins below, default R: gpio5, G: 12, B: 15, W: 13)
//#define WLED_USE_H801 //H801 controller. Please uncomment #define WLED_USE_ANALOG_LEDS as well
//#define WLED_USE_5CH_LEDS  //5 Channel H801 for cold and warm white
//#define WLED_USE_BWLT11
//#define WLED_USE_SHOJO_PCB

#ifndef BTNPIN
#define BTNPIN  0  //button pin. Needs to have pullup (gpio0 recommended)
#endif
```

Legsmaniac, 2020-05-25:
Would still be nice if there was the option to select/change the pin in the UI, perhaps in LED Settings.  Would save having to re-compile/flash.  🙂

bjornbergenheim, 2020-05-25:
Well, how often do you switch around your pins? Would save having to rewrite and introduced more complexity. These are settings that are defined in the config, and lots of places use them in that way. Change it to a runtime settings would be, well a bit of work. To save one recompile and flash. I'm not the guy hosting this code, but I see why it's done this way.

fishbone-git, 2020-05-25:
selectable pins at runtime is on the list for a later release, see #345

Legsmaniac, 2020-05-25:
> Well, how often do you switch around your pins?

Fair point but sometimes I experiment with different boards so it would make life easier.  Once finalised though, like you say, that would be it.  It's set to pin 2 for the data pin but I always opt to use pin 5.  However, there are those who use WLED who can download and flash the bin file but have no idea how to compile.  It would make things easier for newbies for sure.  🙂

@fishbone-git - excellent. Thanks.

jasonblewis, 2020-06-10:
Another issue with Pin 0 is that one of the common ways to drive the 5v addressable LEDs from 3.3v is to use a bidirectional logic level shifter circuit. The input of which usually has a 10k ohm pull up resistor. On GPIO0 this puts the ESP32 into "Normal Execution Mode" rather than ROM serial bootloader which can cause issues with flashing over serial.

The same problem affects GPIO2 which must be left floating or driven low to enter the serial boot loader. Therefore I suggest selecting GPIO other than 0 or 2 if you are going to change the default pin for ESP32.

Example of a level shifter circuit:
https://www.sparkfun.com/products/12009

https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

closed 2020-10-16 by stale[bot]