wled/WLED #937: Even slower twinkle effect :-)

fribse, 2020-05-24:
Hi @Aircoookie 

Thankyou for all the latest inclusions, I have one more request.
I've bought a very thin string of LED's that I plan to use on the christmas tree when it's time for that again,
They look like this https://www.aliexpress.com/item/4000242860388.html?spm=a2g0s.9042311.0.0.27424c4dYBCBgO
I would like for it to just show a constant light, and then at random 'twinkle' one led, but do it very rarely, like a glimmer of a star :-)
I've tried differnet effects, but can't seem to get the right one.
So the effect would be.
Quite dim LED's (single colour) like a regular tree light chain, then one LED goes to full blast very smoothly and short (like a cartoon start twinkle, whole run is max 1 s), then it waits for quite a while (I'm thinking something like two mins or maybe even longer), and then one more does it.
I think it would be a very elegant and discreet way of using this. It would also draw attention, becuase the tree would just be normally lit, and nobody looks at it, and then this twinkle happens, and everybody will look "what was that" and then nothing :-)
I'm not sure if I can accomplish this with the current effects, there are so many of them, but if so, maybe you can point me in the right direction.

Aircoookie, 2020-05-25:
This would be cool to have, I agree!

fribse, 2020-11-18:
I found an example of it, though the 'odd twinkle' should be slower in my opinion, but I guess that could be adjusted with the 'speed'. https://youtu.be/LsCrFiHPo_4

closed 2020-12-12 by fribse