wled/WLED #938: Feature request: Fake TV LED effect

fribse, 2020-05-24:
Hi @Aircoookie 

Now that the summer is coming up, people go on holiday (not much international travel this year though).
A lot of people buy these fake TV led gizmo's that they set up, to portray a living tv, and illustrate that people are home. Most of them are just individually addressable LED's in a matrix of some sort and often in a very poor quality. Maybe this could be a thing for WLED to do?
It wouldn't be that difficult to take a string place it on a platter of some sort, and attach a WLED to it.
Then it could all be controlled like the rest of the light in the house.

Legsmaniac, 2020-05-25:
Just curious but isn't the use of WLED a bit overkill for fake TV?  I mean, isn't this the sort of thing usually created stand-alone, especially since there are already heaps of projects out there for fake TV effects?   🤔

Incidentally, I chanced upon [this](https://learn.adafruit.com/fake-tv-light-for-engineers/overview) very interesting project and just tried it on a spare ESP I had.  Apparently, it's supposed to faithfully reproduce the effect of watching "Wreck It Ralph" and it's certainly very impressive.  🙂

Aircoookie, 2020-05-25:
That project is quite cool, perhaps I am going to implement it as an effect! 25k of data are still quite overkill though, even with 2k of data we should be getting about 10 minutes of non-repeating colors.
WLED is overkill for any one single effect. Just take an Arduino Uno and paste the effect function into an empty sketch. It's power comes from being able to change the "project" at any time just by a phone tap or automation :)

bphick, 2020-05-28:
I had this on my agenda to ask about a while back.  Using such an effect when WLED is already running in soffits / crown molding is a great thing.  Also for "away mode" of the house.  Curtains pulled, TV room looks occupied.

fribse, 2020-06-10:
> Also for "away mode" of the house. Curtains pulled, TV room looks occupied.

Exactly my idea :-)

oz9els, 2020-07-04:
> Just curious but isn't the use of WLED a bit overkill for fake TV? 
isolated: Yes - Probarbly. If you already have your house stuffed with WLED, then it is just a added bonus :-)
So instead of having the WLED changing to a mood when you are home, it changes to the fake-tv, when you are NOT at home.. GENIUS...

Legsmaniac, 2020-07-04:
It's a fair point.
I was going by the many articles I'd read that in order for a "fake-TV" to be realistic, it needs to be created just as we know them, like a small matrix with a diffusing cover over it.  We're not going to be using something like that for mood lighting.  Mood lighting would most likely be strips going round the room, being the TV or created as standing lamps.  I'm not convinced that switching such lighting to the "fake-TV" effect would be realistic enough to serve it's purpose but to be fair, I haven't tried it so I'l willing to be proved wrong.  🙂

Then again, if you already have your house stuffed with nodeMCU's running WLED, what's one more nodeMCU running a custom and stand-along Fake TV, built exactly as it's supposed to be, a small diffused matrix in a specially housed box?  They are so cheap after all. And we're bound to have some spare left-over WS2812B's or similar from our many WLED projects that can be salvaged for a Fake TV.

Still, each to our own I guess, those are only my ideas and thoughts.  😉

sansillusion, 2020-07-05:
I have a strip running in the back of my TV connected to Hyperion. I just
turn off the tv but not the video source so the LEDs keep going as if the
TV was on. Works great since it's real TV signal generating the lighting.

bphick, 2020-07-07:
I have the product named Fake TV.  The original version was weak, and they eventually came out with a brighter one.  While nice, the utility of having it already as a choice in an existing string running WLED would be a huge convenience, vs. dragging out the devices that bear the brand name.

andywarburton, 2020-09-10:
Not sure if this makes your life any easier but I recall the guy from rasp.io wrote example python code for his inspiring range of led strips. It ran super nicely from what I remember so it could well be worth using it as a foundation for this effect in WLED: https://github.com/raspitv/raspio-inspiring/blob/master/TVSimulator.py

fishbone-git, 2020-09-10:
hrm, interesting.  I think random colour changes are the very simple solution but as noted in the first example linked in the comment above, it can still look very fake. I know from personal experience, engineers can get just a bit obsessive, so fake TV for engineers may also be overkill.  Having looked at the video, one should be able to use some fastled math to create some interesting and not obviously repeating cycles of colour, transition time and intensity variation to make an effect that is lean on resources while not being so obvious to anyone not tracking a house so closely they already know the occupants have left.

xxv, 2020-09-11:
So, I actually wrote a TV simulation algorithm that uses FastLED + Arduino. I wrote it for a little night light project that I'm working on which has stalled, but I'd be happy to contribute the code to WLED. I put the source in a gist here: https://gist.github.com/xxv/fcba34658d0d04e7f72f3916e48f87e7

Joshfindit, 2020-11-11:
I wonder if there would be a much simpler way to do this.

We have some known “rules”:
- Stay within an expected palette
- Lots of variation in brightness and saturation around those colours (on the TV this is especially fading in/fading out between scenes and character/scene movement around the frame)
- Mostly gentle (similar colour) transitions, with a few sudden ones being acceptable but not required (like a jump scare in a horror movie)
- A tendency for colours to stay within a certain cluster for 3-15min at a time (scenes)
- Palettes last for no more than 2 hours before changing (22min might even be too short as people typically binge the same show instead of changing shows after each episode)

I suspect that the initial palette could be generated semi-randomly, then the saturations/brightnesses/transitions applied the same to any palette for a realistic TV effect

closed 2020-12-22 by Aircoookie