wled/WLED #940: Night Light Behaviour

Morphy99, 2020-05-24:
I've noticed that if the target brightness is higher than 0 the light will not turn off as stated when clicking the nightlight button in the GUI. I've tested this on the latest 0.10.0 binary using APA102 LED's on a WEMOS D1 mini. I've tried un-ticking Fade Down and it still stays on.

Aircoookie, 2020-05-24:
Hi! This is intended behavior, the nightlight will fade towards the set Target brightness. However you are correct that I should change the wording in the popup, because it indeed does not turn off (and is not meant to) with a target brightness higher than 0.

Morphy99, 2020-05-24:
I thought it would turn off if fade down was unticked? What's the function of fade down then?

Aircoookie, 2020-05-24:
It will turn off if you set the Target Brightness to 0. Fade down will when checked gradually change the brightness over the duration, when unchecked it will stay the same until the duration is over, then instantly apply the set Target brightness :)

commit 4f7dc5be (2020-06-22): Added Sunrise nightlight mode and more UI features

-   Added Sunrise nightlight mode
-   Added Chunchun effect
-   Added `LO` (live override) command to HTTP API
-   Added `mode` to `nl` object of JSON state API, deprecating `fade`
-   Added light color scheme support to web UI (click sun next to brightness slider)
-   Added option to hide labels in web UI (click flame icon next to intensity slider)
-   Added hex color input (click palette icon next to palette select) (resolves #506)
-   Added support for RGB sliders (need to set in localstorage)
-   Added support for custom background color  [... cut]
closed 2020-06-22 by Aircoookie