wled/WLED #942: Feature Request: Include Usermode variables in JSON

chiefymuc, 2020-05-24:
Thats all!

twlare, 2020-05-25:
If you have the ability to recompile the WLED code, you can add the functionality in the `json.cpp` file.

There are two parts to the change. The first is to return the values when a GET request is made. I added the user variables to the `state` object. Sending a GET request to the URL `/json/state` will return the user variable values.
The code to do this should be added to the function `void serializeState(JsonObject root)`. I added it after the `"lor"` object.

```
  root["lor"] = realtimeOverride;

  root["uservar0"] = userVar0;
  root["uservar1"] = userVar1;

  root["mainseg"] = strip.getMainSegmentId();
```

The second change is to set the values when a POST request is made. The user variables are in the `state` object so sending a POST request to the URL `/json/state` will set the user variable values.
The code to do this should be added to the function `bool deserializeState(JsonObject root)`. I added it after the `"lor"` object.

```
  realtimeOverride = root["lor"] | realtimeOverride;
  if (realtimeOverride > 2) realtimeOverride = REALTIME_OVERRIDE_ALWAYS;

  userVar0 = root["uservar0"] | userVar0;
  userVar1 = root["uservar1"] | userVar1;

  byte prevMain = strip.getMainSegmentId();
```

To retrieve the values of the user variables, navigate to `http://<WLED_IP_Address>/json/state`. The response will include the user variable values.
`{'on': true, 'bri': 32, ... 'lor': 0, 'uservar0': 1, 'uservar1': 0, 'mainseg': 0, ...}`

To set the values of the user variables, send a POST request to `/json/state` with new user variable values.
`{'uservar0': 99, 'uservar1': 8888}`

Aircoookie, 2020-05-25:
Thank you @twlare for the excellent explanation :)

I would like to quickly add that I am currently working on a v2 usermod API, which will allow to add values to the state object from the usermod file without having to modify `json.cpp`. Also it will allow multiple usermods at once without having to manually merge them into one file!

Aircoookie, 2020-06-01:
v2 usermods are now implemented!
See https://github.com/atuline/WLED/blob/master/usermods/EXAMPLE_v2/usermod_v2_example.h on how to add usermod variables to JSON :)

closed 2020-06-01 by Aircoookie