wled/WLED #943: Web UI Own Functionality

Legsmaniac, 2020-05-24:
Having a go at trying to change the web UI as per instructions here.

I followed the part where it says
`Change url = command ? '/json/state':'/json'; (0.9.1 line 1048, subject to change) to url = command ? 'http://[Your ESP IP address]/json/state':'http://[Your ESP IP address]/json';`
But the actual code now reads 
`url = rinfo ? '/json/si': (command ? '/json/state':'/json');`
Is this still correct?  Cuz it's not working anyhow.

Although I did discover that if you just load "index.htm" a box pops up asking for the url of a functioning WLED.  I tried entering the URL with and without the "http://" part and no luck either way.
Any help/advice on this?

Aircoookie, 2020-05-25:
Oh, I've indeed changed this to make it easier to run the UI from a file. Just entering the IP in the popup without http:// should be sufficient. Try clearing the local storage (in Chrome go into Developer mode, and in the top right, in the Application tab, you can see the Local storage entries and delete `locIp`). Your WLED installation should be 0.10.0 or newer. What is the error the UI displays?

I will update the docs as the info there no longer applies.

Legsmaniac, 2020-05-25:
That sorted it.  Thanks.

closed 2020-05-25 by Legsmaniac