wled/WLED #945: ESP crashes when changing effects with Peek turned on

Legsmaniac, 2020-05-25:
I think.
I noticed tonight that my ESP's, any of them, appear to crash/freeze/reset or lock up when changing effects on the web UI.  At first I thought it was because I was using segments but I tested another ESP that hasn't got segments set up and that does it too.  Then I thought it was because Preset 16 was saved to. Again, I tested another ESP where no Presets were saved to and still the same problem occurred.  It then dawned on me that the freezes seem to occur if you have Peek turned on while charging and testing various effects.  The red message pops up saying it had lost connection, please refresh or Type error etc etc.  Sometimes the leds freeze, sometimes they carry of running the last selected effect and sometimes they go off altogether.  No amount of refreshing the page brings it back.  I have to reset the ESP.

Anyone else reproduce this error?

Aircoookie, 2020-05-26:
This unfortunately makes sense since the Peek feature sends about 20 requests a second to the ESP. Especially when there are a lot of LEDs connected or the wifi reception isn't perfect (so that requests have to be cached) it can lead to unresponsiveness or even crashes.

I am considering switching to the Websockets protocol both for commands and the peek feature. This could make it more managable as the TCP handshake does not to be done every frame and the ESP can just send the data to the client when it is ready.

Legsmaniac, 2020-05-26:
Ah, ok, good points. 
Thanks for the clarifycation.

Legsmaniac, 2020-05-26:
Further to this.......
I just read the thread in which you uploaded a version of WLED compiled with Core 2.5.2
I decided to download, flash and try this.
I found I am able to rapidly (not that you would normally) change effects with Peek turned on and it never crashed once!
I'm liking this Core 2.5.2 version.  Thinking of recompiling all my boards with 2.5.2 now. 🙃

closed 2020-10-01 by stale[bot]